facebookresearch/habitat-sim #2482: can the sim export the Scenario to other format?

qingchu123, 2024-10-20:
i have find some beautiful scen suit for habitat,is there a method that can export that dataset to other format like a whole glb file so other sim software can use?

aclegg3, 2024-11-19:
Hey @qingchu123 
We don't have any built-in tools for exporting the full scene. 

However, here are a few leads if you'd like to pursue this yourself:

1. You can save a `.scene_instance.json` file with [Simulator.save_current_scene_config](https://aihabitat.org/docs/habitat-sim/habitat_sim.simulator.Simulator.html#save_current_scene_config-6315e). 
2. You can write a Blender script to read scene_instance.json files and load/place all the assets. Then you can export into multiple formats from there.
3. Keep in mind that you will need to use non-compressed assets to do the above. Most Habitat assets are BASIS compressed. Blender and many other engines can't read BASIS. You can try to decompress them with magnum plugins or you can find other source assets which are not in BASIS format.
4. I might suggest that you also have a pre-process script using the MetadataMediator to get the object paths from the datasets instead of trying to do so manually.

qingchu123, 2025-02-03:
can you give some example to use Simulator.save_current_scene_config function to save a .scene_instance.json file? i have download the hssd dataset, but i do not know what to do next