facebookresearch/habitat-sim #2488: ERROR:ImportError: cannot import name 'text' from 'magnum'

WannaSir, 2024-11-03:
when i run the command:
python examples/viewer.py --scene /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb
there will be an error:
Traceback (most recent call last):
  File "/data/nvme2/dky/lc/habitat-sim/examples/viewer.py", line 19, in <module>
    from magnum import shaders, text
ImportError: cannot import name 'text' from 'magnum' (/XX/miniconda3/envs/habitat/lib/python3.9/site-packages/magnum/__init__.py)

Jwonno, 2025-02-11:
I had exactly same errors, did you solve it?

zamo24, 2025-03-21:
@MuXiangWen I am having this issue too. Have you found a fix yet?

fanjing8, 2025-03-27:
i use the [solution ](https://github.com/facebookresearch/habitat-sim/issues/2049#issuecomment-1490568606), but i still have this issue.

2927803072, 2025-04-01:
> i use the [solution ](https://github.com/facebookresearch/habitat-sim/issues/2049#issuecomment-1490568606), but i still have this issue.

@fanjing8 I met the problem too, do you solve it?

fanjing8, 2025-04-02:
> > i use the [solution ](https://github.com/facebookresearch/habitat-sim/issues/2049#issuecomment-1490568606), but i still have this issue.
> 
> [@fanjing8](https://github.com/fanjing8) I met the problem too, do you solve it?

No. I tried to build magnum and magnum-bindings from sources, but all solutions failed.
finally I comment some code related to magnum.text in examples/viewer.py

2927803072, 2025-04-02:
你好，你有遇到过这个问题吗？
Segmentation fault (core dumped)