facebookresearch/habitat-sim #2489: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue

linyx700, 2024-11-06:
[image]

(habitat)` lyx@lyxHP:~/habitat$ python habitat-sim/examples/example.py
sim_cfg.physics_config_file = data/default.physics_config.json
GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidValue

is there anything wrong about the nvidia driver?

Gtrees-w, 2024-11-13:
Hi do you solve this issue?
I met same error.

linyx700, 2024-11-19:
> Hi do you solve this issue? I met same error.

I discovered that my NVIDIA driver had duplicate version conflicts. You can check this.

Eku127, 2024-11-20:
> oh Thanks! I have solved this issue by reinstalling the NVIDIA driver. But I got new isuue when I try to run python examples/interactive_play.py --never-end. X Error of failed request: BadAccess (attempt to access private resource denied) I found the author say this issue haven't solve in Readme.md. Did you meet this issue?

Same issue, any possible solutions? I have followed the instructions in #288 but failed.