facebookresearch/habitat-sim #2524: How to use the semantic ground-truth image segmentation results in HM3D datasets?

zhai-create, 2024-12-29:
## Habitat-Sim version
v0.3.0

When I use `semantic_img = observations["semantic"]` to call the ground-truth image segmentation results, I find that the output result is a matrix of all zeros, which is visualized as a dark image. How can I solve this problem?

## ❓ Questions and Help

zhai-create, 2024-12-29:
Has anyone encountered a similar problem? I am eager to solve this problem, thank you.

zhai-create, 2024-12-30:
It seems that in simulator.py, after calling tgt.read_frame_object_id(self.view), self.view becomes all 0. But why is this?

zhai-create, 2024-12-30:
I still haven't figured this out and I need help.

zamo24, 2025-04-10:
@zhai-create I am seeing this issue too. Have you figured anything out?