facebookresearch/habitat-sim #2534: Segmentation fault (core dumped): OpenGL version not found

Tttt91, 2025-01-18:
## Habitat-Sim version
v0.3.2

## ❓ Questions and Help
I've installed habitat-sim (conda install) and downloaded the dataset.
When I run the Interactive testing command:
`python examples/viewer.py --scene data/scene_datasets/habitat-test-scenes/skokloster-castle.glb`
I encounter the following error, and my OpenGL version is not displayed correctly.
`Platform::GlfwApplication: virtual DPI scaling 1.25
Renderer:  by 
OpenGL version: 
Using optional features:
    GL_ARB_vertex_array_object
    GL_ARB_ES2_compatibility
    GL_ARB_separate_shader_objects
    GL_ARB_robustness
    GL_ARB_texture_storage
    GL_ARB_texture_view
    GL_ARB_framebuffer_no_attachments
    GL_ARB_invalidate_subdata
    GL_ARB_texture_storage_multisample
    GL_ARB_multi_bind
    GL_ARB_direct_state_access
    GL_ARB_get_texture_sub_image
    GL_ARB_texture_filter_anisotropic
    GL_KHR_debug
    GL_KHR_parallel_shader_compile
Using driver workarounds:
    no-layout-qualifiers-on-old-glsl
GL::Context: enabling GPU validation
Segmentation fault (core dumped)`
I can use `glxinfo | grep "OpenGL version"` to check my OpenGL version, and it is:
`OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.2.6`
And `dpkg -l | grep libglvnd-dev`: 
`ii  libglvnd-dev:amd64                          1.3.2-1~ubuntu0.20.04.2               amd64        Vendor neutral GL dispatch library -- development files`
Why is my OpenGL version not displaying?

closed 2025-04-04 by Tttt91

XH-hlx, 2025-07-19:
I have the same error. If you have any solution, please tell me. Thank you.

ZhipengWu1996, 2025-10-30:
I am encountering the same issue as you. May I ask if you are installing habitat-sim on an Ubuntu system within WSL2? Have you resolved this problem?