facebookresearch/habitat-sim #2547: Installation Issues

danieltmeta, 2025-02-14:
## Habitat-Sim version
main

## 🐛 Bug

## Steps to Reproduce

Steps to reproduce the behavior:

```
# Checkout the latest stable release
git clone --branch stable https://github.com/facebookresearch/habitat-sim.git
cd habitat-sim

# We require python>=3.9 and cmake>=3.10
conda create -n habitat python=3.9 cmake=3.14.0
conda activate habitat
pip install -r requirements.txt

# INSTALL HABITAT-SIM
python setup.py build_ext --parallel 6 install --bullet

git clone --branch stable https://github.com/facebookresearch/habitat-lab.git
cd habitat-lab
pip install -e habitat-lab  # install habitat_lab

pip install -e habitat-baselines  # install habitat_baselines

# Testing Data
python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path data/
python -m habitat_sim.utils.datasets_download --uids habitat_test_pointnav_dataset --data-path data/

# Run examples

python ./examples/viewer.py
python ./examples/viewer.py
```

Error Message
```
python setup.py build_ext --parallel 6 install --bullet
/home/trandaniel/dev/habitat-sim/setup.py:432: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  assert StrictVersion(
/home/trandaniel/dev/habitat-sim/setup.py:434: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  ) >= StrictVersion("3.9"), "Must use python 3.9 or newer"
/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/_distutils/dist.py:270: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
running build_ext
CMake Error at CMakeLists.txt:15 (project):
  Running

   '/tmp/pip-build-env-auoxnokv/overlay/bin/ninja' '--version'

  failed with:

   No such file or directory

-- Configuring incomplete, errors occurred!
See also "/home/trandaniel/dev/habitat-sim/build/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
  File "/home/trandaniel/dev/habitat-sim/setup.py", line 441, in <module>
    setup(
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/__init__.py", line 117, in setup
    return distutils.core.setup(**attrs)
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 186, in setup
    return run_commands(dist)
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 202, in run_commands
    dist.run_commands()
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 983, in run_commands
    self.run_command(cmd)
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/dist.py", line 999, in run_command
    super().run_command(command)
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 1002, in run_command
    cmd_obj.run()
  File "/home/trandaniel/dev/habitat-sim/setup.py", line 259, in run
    self.build_extension(ext)
  File "/home/trandaniel/dev/habitat-sim/setup.py", line 346, in build_extension
    subprocess.check_call(
  File "/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '-DBUILD_PYTHON_BINDINGS=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/trandaniel/dev/habitat-sim/build/lib.linux-x86_64-cpython-39/habitat_sim/_ext', '-DPYTHON_EXECUTABLE=/home/trandaniel/miniconda3/envs/hssd_ao_utilities_JT/bin/python', '-DCMAKE_EXPORT_COMPILE_COMMANDS=ON', '-DREL_BUILD_RPATH=ON', '-DCMAKE_BUILD_TYPE=RelWithDebInfo', '-DBUILD_GUI_VIEWERS=ON', '-DBUILD_TEST=OFF', '-DBUILD_WITH_BULLET=ON', '-DBUILD_WITH_CUDA=OFF', '-DBUILD_BASIS_COMPRESSOR=OFF', '-DBUILD_WITH_AUDIO=OFF', '/home/trandaniel/dev/habitat-sim/src']' returned non-zero exit status 1.
```

## Expected behavior

Running the last two commands should generate a window with example scenes to navigate.

danieltmeta, 2025-02-14:
```
# INSTALL HABITAT-SIM
python setup.py build_ext --parallel 6 install --bullet
```
The above step is incorrect!  This needs a --cmake tag:
```
# INSTALL HABITAT-SIM
python setup.py build_ext --parallel 6 install --bullet --cmake
```

closed 2025-02-14 by danieltmeta

BasaErmi, 2025-12-30:
> ```
> # INSTALL HABITAT-SIM
> python setup.py build_ext --parallel 6 install --bullet
> ```
> 
> The above step is incorrect! This needs a --cmake tag:
> 
> ```
> # INSTALL HABITAT-SIM
> python setup.py build_ext --parallel 6 install --bullet --cmake
> ```

thank you very much！🙏 I had been stuck in this problem for a long time until i add a "--cmake"" parameter lol