facebookresearch/habitat-sim #2554: How to Run Habitat Sim and Habitat Lab without a GPU?

mattxgong, 2025-03-13:
Is it possible to run Habitat Sim and Habitat Lab without a GPU? If so, what exact steps are required to do so?

I am running a Ubuntu 18.04 Virtual Machine and was able to successfully install both Habitat Sim and Habitat Lab. However, when I try to run tutorials such as the "Habitat-sim Basics for Navigation" example code , I get this error: Platform::WindowlessEglApplication::tryCreateContext(): unable to find CUDA device 0 among 2 EGL devices in total

McGrady-175, 2025-03-22:
I have the same error, do you solved it?

0mdc, 2025-04-04:
This is not really supported, but there is a hack that may work on your environment.

From `habitat-lab`:

* Try launching with the following config override: `habitat.simulator.habitat_sim_v0.gpu_device_id=-1`.

From `habitat-sim`:

* Set `SimulatorConfiguration.gpu_device_id` to -1.

Let me know if this works for you!

0mdc, 2025-04-07:
>I encounter the problem when I use habitat-sim and I follow your instruction and finally it works!!!

Awesome! Thanks for trying it out!

I'll figure out a way to make this hack documented and officially supported.

LightwishWONG, 2025-04-10:
> > This is not really supported, but there is a hack that may work on your environment.
> > From `habitat-lab`:
> > ```
> > * Try launching with the following config override: `habitat.simulator.habitat_sim_v0.gpu_device_id=-1`.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > From `habitat-sim`:
> > ```
> > * Set `SimulatorConfiguration.gpu_device_id` to -1.
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Let me know if this works for you!
> 
> I encounter the problem when I use habitat-sim and I follow your instruction and finally it works!!!

Hi. Could you please tell me where exactly to set those parameters? I'm new to habitat and also looking for running habitat without GPU. Many thanks!

LightwishWONG, 2025-04-10:
> > > > This is not really supported, but there is a hack that may work on your environment.
> > > > From `habitat-lab`:
> > > > ```
> > > > * Try launching with the following config override: `habitat.simulator.habitat_sim_v0.gpu_device_id=-1`.
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > From `habitat-sim`:
> > > > ```
> > > > * Set `SimulatorConfiguration.gpu_device_id` to -1.
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > Let me know if this works for you!
> > > 
> > > 
> > > I encounter the problem when I use habitat-sim and I follow your instruction and finally it works!!!
> > 
> > 
> > Hi. Could you please tell me where exactly to set those parameters? I'm new to habitat and also looking for running habitat without GPU. Many thanks!
> 
> I am new to habitat too. In my project, I just set the gpu_device_id to -1 after Instantiate a SimulatorConfiguration object. This is my part of `code:# simulator backend sim_cfg = habitat_sim.SimulatorConfiguration() sim_cfg.gpu_device_id=-1#######this line was added by likelai sim_cfg.scene_id = settings["scene"] sim_cfg.scene_dataset_config_file = settings["scene_dataset_config_file"] sim_cfg.load_semantic_mesh = True`

Thanks for your quick reply! Here which file did you add this line? I've installed habitat-sim by conda install, and I'm running the example/viewer.py to visualize. I searched sim_cfg = habitat_sim.SimulatorConfiguration() but there are many files containing it.

aclegg3, 2025-04-10:
@LightwishWONG in general you should inject this configuration change directly before the Simulator object is instantiated in whatever script or application you are running. In viewer.py you can inject it here: https://github.com/facebookresearch/habitat-sim/blob/2aa5ed12c639e02dc19ea3b5fd2df7763ef176af/examples/viewer.py#L389 just before the first Simulator object is created.