facebookresearch/habitat-sim #2561: ‘GL::Context: cannot retrieve OpenGL version’ Problem

Eku127, 2025-04-03:
## Habitat-Sim version
v0.3.3 Stable branch

## Docs and Tutorials
Did you read the docs? Y

Did you check out the tutorials? Y

## ❓ Questions and Help
Hello, I am currently trying to set up habitat-sim under a Python 3.10 environment.

I followed the instructions from the official documentation at [here](https://github.com/facebookresearch/habitat-sim/blob/main/BUILD_FROM_SOURCE.md)

Specifically, I cloned the repository using the stable branch, and installed it with the following command:
```
CMAKE_ARGS="-DCMAKE_POLICY_VERSION_MINIMUM=3.5" pip install . -v --config-settings editable=true
```
The build process completed successfully, but when I tried to run my application, I encountered the following OpenGL-related error `GL::Context: cannot retrieve OpenGL version`

I came across this [merged PR](https://github.com/facebookresearch/habitat-sim/pull/2519) that seems related to my issue.
Following the proposed changes in that PR, I manually remove imageio-ffmpeg. However, the same error still occurs.

My questions are:

- Does the fix proposed in that PR apply to builds from source, or only to prebuilt binaries?

- Are there any additional steps I should take to resolve this OpenGL context issue when compiling manually?

I have attached the current conda environment information as an image. If more information is needed, I’m happy to provide it. Thank you for your time and help. @0mdc 

[env.txt](https://github.com/user-attachments/files/19582360/env.txt)

Eku127, 2025-04-04:
Problem Solved.

closed 2025-04-04 by Eku127

0mdc, 2025-04-04:
Hi @Eku127 !

What worked for you?

Eku127, 2025-04-07:
> Hi [@Eku127](https://github.com/Eku127) !
> 
> What worked for you?

I'm not an expert in this area, and I ended up solving the problem through some trial and error.

Initially, running `pip install . ` caused GL error to appear. I then tried running `pip install -e .` afterward, which resolved the issue.

However, if I solely run `pip install -e .` first, I get a different error related to a missing dependency (magnum).

Here’s the exact script I used:
```
echo "[1/3] Installing habitat-sim (editable mode)"
CMAKE_ARGS="-DCMAKE_POLICY_VERSION_MINIMUM=3.5" \
pip install . -v
CMAKE_ARGS="-DCMAKE_POLICY_VERSION_MINIMUM=3.5" \
pip install -e .
```

It's definitely not an ideal or elegant solution. I'm sharing this in case it helps others, and I’d really appreciate it if someone could explain why this happens =.=

Eku127, 2025-04-16:
> Hi [@Eku127](https://github.com/Eku127) !
> 
> What worked for you?

https://github.com/Eku127/habitat-data-collector/blob/main/scripts/install_habitat.sh

That is my final solution =-=

0mdc, 2025-04-16:
Thanks for the update!

We'll update the build instructions for the `pip` flow.
`pip install` without `-e` probably doesn't play well with the build scripts - I suspect that you would need to re-do the command every time to update the build without the `-e` parameter.

The build scripts aren't following the recent Python standards, it'll be good to review them.