facebookresearch/habitat-sim #2562: Failed to get EGL display

shuowang178, 2025-04-03:
EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate(): cannot create a window with core OpenGL context, falling back to compatibility context
EGL: Failed to get EGL display: Success
Platform::GlfwApplication::tryCreate(): cannot create a window with OpenGL context

May I ask how to solve this problem? Could you please tell me that this problem has been bothering me for a long time, thank you very much.

shuowang178, 2025-04-05:
hi,
I've already tried the linux section using the URL you sent.Please see troubleshooting procedure.Please help me solve this problem, which has been bothering me for a long time.Thank you.

 

 

 

 

 

 

 

  

 

 

---- Replied Message ----
| From | Mikaël Dallaire ***@***.***> |
| Date | 4/5/2025 00:04 |
| To | ***@***.***> |
| Cc | ***@***.***>,
***@***.***> |
| Subject | Re: [facebookresearch/habitat-sim] Failed to get EGL display (Issue #2562) |

Hi @shuowang178

This looks like a driver or environment issue.

What is your OS and GPU?

Check out this document for some troubleshooting steps: https://github.com/facebookresearch/habitat-lab/blob/main/TROUBLESHOOTING.md#unable-to-create-context

—

Hi @shuowang178

This looks like a driver or environment issue.

What is your OS and GPU?

Check out this document for some troubleshooting steps: https://github.com/facebookresearch/habitat-lab/blob/main/TROUBLESHOOTING.md#unable-to-create-context

shuowang178, 2025-04-05:
> Hi [@shuowang178](https://github.com/shuowang178)
> 
> This looks like a driver or environment issue.
> 
> What is your OS and GPU?
> 
> Check out this document for some troubleshooting steps: https://github.com/facebookresearch/habitat-lab/blob/main/TROUBLESHOOTING.md#unable-to-create-context
hi@0mdc 
I've already tried the linux section using the URL you sent.Please see troubleshooting procedure.Please help me solve this problem, which has been bothering me for a long time.Thank you.

[新建文件夹.zip](https://github.com/user-attachments/files/19614726/default.zip)

InkDeying, 2025-06-15:
> [新建文件夹.zip](https://github.com/user-attachments/files/19614726/default.zip)

你好，问题解决了吗？