facebookresearch/habitat-sim #2563: Problem about semantic scene

DuoChen-317, 2025-04-09:
Hi, i try to use semantic scene to get the region label according the position, the position is from the navmesh, but it just return nothing when i use def get_regions_for_point. I have already load the house file using def load_mp3d_house. FYI, it still return nothing when i random pick a center node from on region aabb, but still return nothing, but after i use my own function(just check if the position is in the aabb) it work but still return on the sampling position

DuoChen-317, 2025-04-09:
the sampling position is from the function pathfinder.get_random_navigable_point()

closed 2025-04-09 by DuoChen-317

shuitatata, 2025-08-06:
Hi there!
I’m encountering the same issue in Matterport3D scenes: `scene.get_regions_for_point(point)` always returns an empty list, even when sampling points known to be inside a target region. I verified `semantic_scene.load_mp3d_house(...)` is enabled.

Could you please share how you eventually solved it?

Any insight into how you determined region membership from navmesh points would be hugely helpful. Thanks!

DuoChen-317, 2025-08-07:
Hi, i solved the problem long time ago and i don’t really how the details works. But I do remember some possible solutions, for each scene you can check the aabb to see if the sampling point is matched. cause i kind of remember the some of the sampling points are mismatched. I hope that can help you. And I can check codes for more details if you want