facebookresearch/habitat-sim #2564: skokloster-castle.scn` does not exist.

behappya, 2025-04-17:
[10:07:35:876687]:[Warning]:[Scene] SemanticScene.h(331)::checkFileExists : ::loadSemanticSceneDescriptor: File `./data/scene_datasets/habitat-test-scenes/skokloster-castle.scn` does not exist.  Aborting load.
[10:07:35:876695]:[Warning]:[Scene] SemanticScene.cpp(123)::loadSemanticSceneDescriptor : SSD File Naming Issue! Neither SemanticAttributes-provided name : `./data/scene_datasets/habitat-test-scenes/skokloster-castle.scn` nor constructed filename : `./data/scene_datasets/habitat-test-scenes/info_semantic.json` exist on disk.
[10:07:35:876700]:[Error]:[Scene] SemanticScene.cpp(139)::loadSemanticSceneDescriptor : SSD Load Failure! File with SemanticAttributes-provided name `./data/scene_datasets/habitat-test-scenes/skokloster-castle.scn` exists but failed to load.
[10:07:38:026267]:[Warning]:[Sim] Simulator.cpp(595)::instanceStageForSceneAttributes : The active scene does not contain semantic annotations : activeSemanticSceneID_ = 0

LiuXingZhi11, 2025-05-22:
hello, I also met the same problem. Did you solve it?

alwualwu, 2025-12-27:
> [10:07:35:876687]:[Warning]:[Scene] SemanticScene.h(331)::checkFileExists : ::loadSemanticSceneDescriptor: File `./data/scene_datasets/habitat-test-scenes/skokloster-castle.scn` does not exist. [...]

Hello, may I ask if you can solve this problem