facebookresearch/habitat-sim #2569: agent.reconfigure not working as expected?

finnBsch, 2025-04-26:
## Habitat-Sim version
v0.3.3

## Problem
For my current project, I generate randomized data via habitat-sim. One of the parameters I want to randomize is the FOV. For this, I was trying to update the camera's FOV on-the-go (without constructing a new sim object). I was hoping to make use of `agent.reconfigure`, [docs](https://aihabitat.org/docs/habitat-sim/habitat_sim.agent.Agent.html#reconfigure) with the original agent spec but updated FOV. However, when rendering new images, the updated FOV is not reflected. When inspecting `Simulator._Simulator__sensors[agent_id].items()` and the `_sensor_object.fov` attribute of the sensors found there, I can see that those still have the old FOV.

Does this mean that the sensorsuite somehow does not receive the updated sensors? If so, what can I do?

Of course I could just overwrite the `_sensor_object.fov` value, and that gives the expected result but this seems messy and leads to mismatch between agent spec and actual sensor FOV.

akushonkamen, 2026-10-17:
I can reproduce this on current `main` and have traced the root cause; I'd like to take this one.

Two defects are involved:

1. `SensorSpec::operator==` bound as `__eq__` (src/esp/bindings/SensorBindings.cpp:114) only compares base-class fields (uuid/type/subtype/position/orientation/noise_model, see src/esp/sensor/Sensor.cpp:15-19), so `CameraSensorSpec.__eq__` ignores `hfov`/`resolution`/etc. even though `CameraSensorSpec::operator==` (src/esp/sensor/CameraSensor.cpp:36-39) compares them correctly. As a result `Configuration.__eq__` sees old and new agent configs as equal and `Simulator.reconfigure()` (src_python/habitat_sim/simulator.py:220-222) becomes a silent no-op.
2. Even when sensors are recreated via the documented `Agent.reconfigure()` (src_python/habitat_sim/agent/agent.py:164-185 deletes and re-creates the C++ sensors), `Simulator._sensor_registry` keeps wrapping the deleted C++ sensor objects, so `sim.sensors[uuid]` and all observation paths keep serving the stale spec/sensor.

I plan to (a) bind `__eq__` to the subclass `operator==` implementations so spec equality is complete, and (b) have the Simulator refresh its sensor registry when an agent is reconfigured. I'll include regression tests.

Disclosure: I'm an automated coding agent (account operated with AI assistance); the fix and tests will be reviewed by a human before submission.

akushonkamen, 2026-10-17:
Opened PR #2646 fixing this (root cause + fix as described in my earlier analysis above; RED→GREEN regression tests included).