facebookresearch/habitat-sim #2576: GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidOperation

missTL, 2025-06-25:
How to solve the habitat-sim error：
GL::Context: cannot retrieve OpenGL version: GL::Renderer::Error::InvalidOperation
Aborted

evelinehong, 2025-09-09:
similar problem with H200

RoseWillow, 2025-11-09:
> I had exactly the same problem and managed to solve it. The issue is that the system is trying to use Mesa's software rendering libraries instead of the NVIDIA GPU libraries. You should replace the Mesa EGL libraries with the NVIDIA ones by creating symlinks. If you need more details, let me know.

Yes，more details，please！

yourkln, 2025-11-09:
> Yes，more details，please！

@RoseWillow please run this and let me know what was your output: python -c "import habitat_sim; sim_cfg = habitat_sim.SimulatorConfiguration(); print('Habitat-Sim initialized')"

RoseWillow, 2025-11-09:
> [@RoseWillow](https://github.com/RoseWillow) please run this and let me know what was your output: python -c "import habitat_sim; sim_cfg = habitat_sim.SimulatorConfiguration(); print('Habitat-Sim [...]

Thanks for your reply, I have run ''export LD_PRELOAD=/lib/x86_64-linux-gnu/libGLX_nvidia.so.0:/lib/x86_64-linux-gnu/libGLdispatch.so.0'' and solve my problem. Thanks again!