facebookresearch/habitat-sim #2578: libEGL warning: DRI2: failed to authenticate     Segmentation fault (core dumped)

XH-hlx, 2025-07-19:
[image]
"habitat-viewer /path/to/data/scene_datasets/habitat-test-scenes/skokloster-castle.glb"

I'm new to this, I don't know why it's wrong, please help me solve this problem. Thanks

Jwonno, 2025-07-21:
Hi, did you solved it? I' faced wiht same error...

singhsegv, 2025-07-23:
Same error

JHY491, 2025-07-23:
Hi. just change the /path to your absolute path in systems.

XH-hlx, 2025-07-23:
> 你好，你解决了吗？我也遇到了同样的错误……

> 同样的错误

When configuring the environment, you need to “conda install” the corresponding version of habitat-sim downloaded. 
https://anaconda.org/aihabitat/habitat-sim/files

Jwonno, 2025-07-28:
> > 你好，你解决了吗？我也遇到了同样的错误……
> 
> > 同样的错误
> 
> When configuring the environment, you need to “conda install” the corresponding version of habitat-sim downloaded. https://anaconda.org/aihabitat/habitat-sim/files

What does "corresponding version of habitat-sim" mean? What is it supposed to correspond with?

I followed the general scenario and used the following command to install it in my conda environment:
`conda install habitat-sim withbullet -c conda-forge -c aihabitat`

XH-hlx, 2025-07-30:
> What does "corresponding version of habitat-sim" mean? What is it supposed to correspond with?
>
> I followed the general scenario and used the following command to install it in my conda environment: [...]

I mean that  download the "habitat-sim‘s .conda" file and conda install it in your terminal, instead of using the "-c conda-forge ..."

Jwonno, 2025-07-30:
@XH-hlx I see. So I should keep trying 'conda install' with different "habitat-sim's .conda" file until it works. is it right?

marcelprasetyo, 2025-08-29:
I got this error on two separate computers and solved it. This is probably an issue with the nvidia driver or libraries from the driver packages. It may also be an issue with recent conda/pip package versions which may be related to what @XH-hlx pointed out. See my more detailed answer for [#2583](https://github.com/facebookresearch/habitat-sim/issues/2583#issuecomment-3235670961) for a conda environment that works. For nvidia driver, try upgrading / downgrading to nvidia-driver-550.